facebookresearch/digit-design #1: digit vs digit-flex

b4be1, 2020-05-31:
Dear digit-design team,

I want to fabricate the digit sensor. In the `hardware` directory, there are two options: `digit` and `digit-flex`. Are these two different version of the sensor or are these different parts of the same sensor and I need to fabricate them both?

b4be1, 2020-06-01:
Ok, I think `digit` refers to the camera PCB (Fig. 2F in the [paper](https://arxiv.org/pdf/2005.14679.pdf)) and `digit-flex` refers to the lighting PCB (Fig. 2D). So, one needs to manufacture both.

I am trying to submit an order with https://www.pcbway.com/, but there are many options that need to be specified for fabrication which are not stated in the paper. It would be nice if you provide instructions on how to fabricate the sensor.

b4be1, 2020-06-01:
For example, is `digit` a 2-layer board or 4-layer? There are files
```
digit_copper_signal_1.gbr
digit_copper_signal_2.gbr
digit_copper_signal_bot.gbr
digit_copper_signal_top.gbr
```
Does this mean it is a 4-layer board? In which order should these layers go?

AlphaBetaPhi, 2020-06-01:
Hi @b4be1, 

We will be releasing a quick-start guide outlining the assembly process. Please see the image below for manufacturing specifications.

[image]

The DIGIT top level assembly contains two PCB's, the main 4 layer PCB and the flex lighting PCB.

AlphaBetaPhi, 2020-06-01:
Hi @b4be1,

We have uploaded the quick start guide which contains more detailed information for the manufacturing and assembly of the device.

https://github.com/facebookresearch/digit-design/blob/master/DIGIT_Quick_Start_Guide_v1.pdf

robertocalandra, 2020-06-01:
This should answer the question :) Closing the issue. feel free to open a new one if you have more questions.

closed 2020-06-01 by robertocalandra