facebookresearch/digit-design #2: Microcontroller FT900Q-R is obsolete. Should it be replaced with FT900Q-C-R?

b4be1, 2020-06-02:
It seems the FTDI microcontroller FT900Q-R is not available anymore, see e.g. https://www.mouser.de/ProductDetail/Bridgetek/FT900Q-R?qs=cGEy3R83DS9XW6la3MeywQ==

It should probably be replaced with the newer version FT900Q-C-R. I guess that should work, right?

AlphaBetaPhi, 2020-06-02:
Hei @b4be1,

The DIGIT is based off of the FT900Q series, silicon revision C. At the time of design engineering samples of silicon revision C were available through the FT900Q-R part number. This is equivalent to FT900Q-C-R.

b4be1, 2020-06-02:
Got it, thanks!

closed 2020-06-02 by b4be1

nouyang, 2020-08-04:
In case of use; per datasheet for FT900 [e.g here]( https://brtchip.com/wp-content/uploads/Support/Documentation/Datasheets/ICs/MCU/DS_FT900_1_2_3.pdf).
 FT900Q-C-X --> the last letter indicates only product packaging and should be interchangeable. 
```
-R: Tape and Reel(Qty per reel is 1000)
-T: Tray packing(Qty per tray for LQFP is 90, qty per tray for QFN is 152
```