facebookresearch/digit-design #3: How to get the gel

callmeray, 2020-06-05:
Hi,

Thank you for your excellent work and sharing.

I'm trying to fabricate one DIGIT and I've read the quickstart. But there is little information about the gel. Where can I get the gel? Or do I need to make it myself?

Thanks again!

robertocalandra, 2020-06-05:
Hi Rui,
Thank you for your interest.
The elastomer can not, unfortunately, be bought anywhere since it is very specialized, so you will have to manufacture the elastomer on your own.
The general information for manufacturing the elastomer is in the [paper](https://arxiv.org/abs/2005.14679), but we aim to produce a more detailed guide specifically for elastomer manufacturing in the near future.

callmeray, 2020-06-05:
Thanks for your quick reply. I'm looking forward to seeing the guide.

closed 2020-06-05 by callmeray

AlphaBetaPhi, 2020-06-05:
Hello @callmeray,

The quick start guide has been updated to include the gel manufacturing process.

callmeray, 2020-06-06:
Thanks a lot! I'll read it and try to manufacture it.