facebookresearch/digit-design #5: Rough cost of electronics?

nouyang, 2020-06-15:
Hello! 
Thanks for the release of [all the files](https://github.com/facebookresearch/digit-design/blob/master/releases/DIGIT-fab-release-2020-1.zip) needed to contact a PCB manufacturer for the boards and assembly. I see also there is a [bill of materials](https://github.com/facebookresearch/digit-design/blob/master/hardware/Digit%20BOM%20-%20rev%202020-1.pdf) and a nice [quick start guide](https://github.com/facebookresearch/digit-design/blob/master/DIGIT_Quick_Start_Guide.pdf) It's an amazing amount of work and documentation to be made available to the tactile sensing and robotic manipulation community. I wonder if it's possible to also get a rough sense of the costs at different quantities - e.g. for quantity 5, 10, 50, 100. Apologies as I have only ordered bare PCBs before, this will help me understand whether the cost is reasonable for us to look into. Thanks!

AlphaBetaPhi, 2020-06-15:
Hello @nouyang,

Thank you for your interest and kind words. The pricing entirely depends on the 3rd party manufacturer, assembly type, lead times for the components and PCB's.

If you're looking for an accurate cost figure, I would recommend supplying the release package to various vendors for quoting in those quantities (5, 10, 25, 100, etc). Additionally, if you're looking at hand soldering and assembly, the estimated cost figure for a quantity of 10 is $60-$80 per device. This declines rapidly based on the number of devices being built.

AlphaBetaPhi, 2020-06-16:
Closing this, feel free to re-open or create a new issue.

closed 2020-06-16 by AlphaBetaPhi

nouyang, 2020-06-16:
Yes, my apologies for being unclear - I meant that I have not spec'd boards before where the manufacturer also populated the components onto the board, which seems to be recommended in this case. Thank you!

AlphaBetaPhi, 2020-06-16:
Hi @nouyang, 

With the manufacturing release package you should be able to get an accurate cost figure for various quantities with automated assembly.  Hope this helps :)

nouyang, 2020-06-17:
Thanks! Yes, I struggled a bit to submit for a quote since it was/is a bit difficult for me to figure out straight-away e.g. number of layers the board has, number of SMT pads, etc.
To clarify, when you say "if you're looking at hand soldering and assembly, "  - I'm actually not sure what that means. Slowly chugging along this process but any tips appreciated :)

nouyang, 2020-07-29:
In case anyone is interested, my quote from pcbway was ~$90 in qty 20 and going down to ~$80 at qty 55. I've made a sheet at tinyurl.com/digit-order & am submitting today. I will look to see if there is option to have circuitboard easily order-able, as they have a "open source" option.