facebookresearch/digit-design #7: Layout of LED

lzyplayer, 2020-06-30:
Hi! 
Thanks for opening source such great work!

I'm wondering how the rgb light layout on the surface is designed.
[image]
It seems blue light cover the whole image, while red and grenn only cross a little compared with gelsight LED layout.
[image]
Will it be a problem on reconstructing the surface?

Thanks for any reply!

lzyplayer, 2020-07-01:
Another question, 
[image]
the quantities doesn't match current designator. Also, it seems 5 LED is required accroding to [Digit Flex Schematic - rev 2020-1.pdf](https://github.com/facebookresearch/digit-design/blob/master/hardware/Digit%20Flex%20Schematic%20-%20rev%202020-1.pdf)
Thanks for any reply!

AlphaBetaPhi, 2020-07-01:
Hi @lzyplayer,

Thank you for your feedback and interest.

Up to 4 LED's may be included, however, for DIGIT we only populate D1, D2, D3. The compressed release file specifies the populated components in "digit_flex_smt_xy.txt".

I will update the BOM PDF to state that these are optional.

There will be an update to this issue regarding your the initial comment about the lighting configuration.

lzyplayer, 2020-07-01:
Thanks for your quick reply! That solve my latter question. 
Really looking forword to the upcomming update!

AlphaBetaPhi, 2020-07-03:
Hi @lzyplayer,

The DIGIT was designed to resemble that of a finger tip. Two LED's (red and green) are offset by 45deg from the center, and the blue LED is directly in the center. I do believe it might impact 3D reconstruction performance but this is something we do not have any numbers to verify this.

closed 2020-07-03 by AlphaBetaPhi

lzyplayer, 2020-07-03:
Thanks for your kindly reply! I 'll try some reconstruction work on current DIGIT sensor and share what I found.