facebookresearch/digit-design #10: Acrylic_Laser.DXF

vbetancu, 2020-07-31:
Good afternoon,
I have found you work to be very interesting, thanks for sharing.
While trying to open the files I was unable to open the acrylic_laser.dxf. It shows as text but doesn't work with AutoDesk software. Could you please help me?
Thanks

AlphaBetaPhi, 2020-07-31:
Hi @vbetancu,

Thank you for your interest in this work!

Do you have a preferred file format?

vbetancu, 2020-07-31:
I am trying to open it with AutoCad or AutoDesk Fusion, so I believe DXF is good. I am just not sure why it opens as text and can not be read by these software packages.

vbetancu, 2020-08-03:
Maybe if you could help me by re-uploading the file. That might fix the issue. Thanks in advance.

vbetancu, 2020-08-03:
I also had a similar issue with the Gel mold file. AutoDesk was unable to read it.

robertocalandra, 2020-08-03:
Hi @vbetancu,
We tested acrylic_laser.dxf on several machines/softwares (including AutoDesk Fusion 360, and the online viewer https://beta.sharecad.org) and we could not reproduce this issue. For us, everything seems to be working correctly. 
I suspect that it might be an issue with your local machine. What OS are you using? can you successfully open other .dxf files?

PS: say Hi to Kimo from me :)

closed 2020-08-03 by robertocalandra

vbetancu, 2020-08-04:
Hi @robertocalandra ,
Thank you for your message. I'll tell Kimo you say hi :)
I think my issue might be with the download of the file as when I click on it on the webpage there is no option to download like the others but it translates into text. Do you recommend any way of downloading it directly from the page? That might solve my issue.

robertocalandra, 2020-08-04:
The best way to download the files is either to manually download the whole github repo from https://github.com/facebookresearch/digit-design/archive/master.zip or (preferably) to git clone the repo. On the main page of the repo (https://github.com/facebookresearch/digit-design) you should see a green button named "Code" that lets you explore these two operations.

vbetancu, 2020-08-04:
Thank you @robertocalandra, that actually solved my problem. I really appreciate your help!

vbetancu, 2020-08-06:
Hi @robertocalandra, from the gel mold I see that you have some holes in the corners. Could I ask you what the purpose of these holes was?

AlphaBetaPhi, 2020-08-06:
Hi @vbetancu,

The holes included in the gel mold are optional for molding low quantities. We have included them as locator holes for mounting plates or an optional self-designed back piece.

vbetancu, 2020-08-07:
Thanks a lot Mike!