facebookresearch/digit-design #11: Cannot set serial correctly

callmeray, 2020-08-04:
Hi,

Thank you for your great work.

I tried to flash the firmware with a serial number, but got "DXXXXX" when connected it using [https://github.com/facebookresearch/digit-interface](https://github.com/facebookresearch/digit-interface)

AlphaBetaPhi, 2020-08-04:
Hi @callmeray,

Thank you for your interest!

Can you provide the command you used to flash the firmware?

callmeray, 2020-08-05:
Thank you for your reply. 

The command was `python3 flash.py digit.serial=45`. I think it is more like the interface issue, since I could generate the temporary firmware.

AlphaBetaPhi, 2020-08-05:
Hi @callmeray, can you run `lsusb -v` and see if it reports the correct serial number there?

callmeray, 2020-08-05:
The result is 

> Bus 003 Device 021: ID 2833:0209  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x2833 
  idProduct          0x0209 
  bcdDevice            1.01
  iManufacturer           1 Facebook
  iProduct                2 DIGIT
  iSerial                 3 DXXXXX
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          220
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass         14 Video
      bFunctionSubClass       3 Video Interface Collection
      bFunctionProtocol       0 
      iFunction               2 DIGIT
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      1 Video Control
      bInterfaceProtocol      0 
      iInterface              2 DIGIT

AlphaBetaPhi, 2020-08-05:
@callmeray,

Thank you, looks like the serial is reported as `iSerial 3 DXXXXX`.

Can you `git pull` from the repo and then run this command,

`python3 flash.py digit.firmware="../digit-2020-1b.bin" digit.serial=45`

callmeray, 2020-08-05:
I got the same result.

>  idVendor           0x2833 
  idProduct          0x0209 
  bcdDevice            1.01
  iManufacturer           1 Facebook
  iProduct                2 DIGIT
  iSerial                 3 DXXXXX

Does it require a specified python version? I'm using Python 3.7.7.

AlphaBetaPhi, 2020-08-05:
@callmeray,

May you post the output of `flash.py`?

callmeray, 2020-08-06:
Thanks. It works this time. The reason I failed should be that DIGIT was plugged when I run `flash.py`.

robertocalandra, 2020-08-06:
Glad to hear that this fixed the problem. I am closing the issue; feel free to reopen if you encounter more problems.

closed 2020-08-06 by robertocalandra