facebookresearch/digit-design #13: U2 (OVM7692-RAAA) not available to purchase

robotfan32123, 2020-09-22:
Hi Everyone! First thing first: Great Project!
We'd love to build some test units of the digit but unfortunately found that its impossible to source U2 (OVM7692-RAAA) over here in Germany. We tried every supplier we know but only find it in the US (export restrictions don't allow digikey to ship it here from their US facilities). OmniVision themself only start selling >10k units/year.

Are you aware of this issue and do you maybe have a solution/replacement for this?

Thanks in advance!

AlphaBetaPhi, 2020-09-22:
Hi @robotfan32123, thank you for your interest in the project!

Thank you for bringing this up. Currently, there is no replacement to the OVM7692 sensor.

However, it is recommended to build this through a fabrication house due to the fine pitched/BGA components on the PCB. The fabrication release package may be sent to a number fabrications house for builds which will use their established supply chains, have you tried this?

robotfan32123, 2020-09-22:
Thanks for the quick reply, @AlphaBetaPhi 
Yes, we tried 2 different EMS' we usually comission for fabrication (one of them being quite big here in Germany) but both of them struggle to source the sensor.
They even tried to alternatively source from a HK based supplier or directly contacted OmniVision but had no success (unless we order >10k units.)

Is it possible to DM you for this? Maybe we can find a different way of sourcing the components or assembled PCBs.

AlphaBetaPhi, 2020-09-22:
Hi @robotfan32123,

Feel free to reach out offline at lambetam [at] fb.com

Qianboy, 2021-01-07:
Hi all,

We are also super willing to try this sensor out here in Germany, since we are also facing this issue, can you please introduce a bit on how to source the missing components? Thanks a lot!

AlphaBetaPhi, 2021-01-08:
Hi @Qianboy,

May you reach out to me at lambetam[at]fb.com

Thank you,

Mike

robertocalandra, 2021-01-08:
We are aware that unfortunately sourcing the OVM7692 can be complicated in some countries, but we do not plan to support other cameras at the time being. 
I am closing the issue, but for anybody encountering this issue now and in the future, please get in touch directly with @AlphaBetaPhi at lambetam[at]fb.com and we will try our best to assist.

closed 2021-01-08 by robertocalandra