facebookresearch/digit-design #15: Programming Digit Device

vbetancu, 2020-11-04:
Good evening,

I was trying to flash the firmware for the Digit device but it seems like the instructions (at least for the programmer) are for Linux. Do you have instructions for Windows by chance?

Thanks in advance,
Veronica

AlphaBetaPhi, 2020-11-05:
Hello @vbetancu,

Thank you for your interest. 

You may follow the link in this post to download the toolchain for Windows.

https://github.com/facebookresearch/digit-interface/issues/3#issuecomment-657252046

vbetancu, 2020-11-05:
Thank you for your response. I followed the instructions from the other post and downloaded the toolchain and flashed "digit-2020-1b.bin" to the device.
Then I went to the programmer and changed the serial to D00001.
What would be the next step?

vbetancu, 2020-11-05:
I unplugged and plugged back device and now it's recognized. Then I followed installation instructions in this link:https://github.com/facebookresearch/digit-interface/blob/master/README.md
How to proceed for the usage section? Spyder or similar?

AlphaBetaPhi, 2020-11-09:
Hi @vbetancu,

I have attached a picture for future reference on Windows programming.

[image]

The DIGIT interface currently only supports Linux based operating systems. However, to use DIGIT on Windows you can initialize the DIGIT class without a serial number and then populate the device name using the OpenCV device ID (in this example it is 0):

```
from digit_interface.digit import Digit
 
d = Digit(name="DIGIT Dev 0")
d.dev_name = 0

d.connect()
d.show_view()
d.disconnect()
```

vbetancu, 2020-11-18:
Thank you very much Mike. 
I was able to get the device to stream video but it's not able to take single frames. I am following the procedure in the read me file. Am I missing something?

from digit_interface.digit import Digit
d = Digit("DXXXXX") # Unique serial number
d.connect()
frame = d.get_frame()
print(f"Frame WxH: {frame.shape[0]}{frame.shape[1]}")

AlphaBetaPhi, 2020-11-18:
Hi @vbetancu,

We do not support connecting to the serial number of the device on Windows platforms. Please try the following,

```
from digit_interface.digit import Digit
 
d = Digit(name="DIGIT Dev 0")
d.dev_name = 0
d.connect()

frame = d.get_frame()
print(f"Frame WxH: {frame.shape[0]}{frame.shape[1]}")
```

vbetancu, 2020-11-18:
Thank you Mike. I actually switched to a Linux operating system when trying to implement the code in the previous message.

AlphaBetaPhi, 2020-11-18:
What does the output show?

vbetancu, 2020-11-18:
runfile('/home/gelsight/.config/spyder-py3/temp.py', wdir='/home/gelsight/.config/spyder-py3')
Frame WxH: 320240

But it didn't take a picture

vbetancu, 2020-11-18:
That's the output I get after running: 
from digit_interface.digit import Digit
d = Digit("DXXXXX") # Unique serial number
d.connect()
frame = d.get_frame()
print(f"Frame WxH: {frame.shape[0]}{frame.shape[1]}")

AlphaBetaPhi, 2020-11-18:
`d.get_frame()` returns a frame into the variable `frame` which can then later be used for processing in the application. This does not save the frame. To do this, 

```
from digit_interface.digit import Digit
d = Digit("DXXXXX") # Unique serial number
d.connect()
frame = d.save_frame()
```

vbetancu, 2020-11-18:
That worked! Thank you MIke!

AlphaBetaPhi, 2020-11-18:
Great to hear @vbetancu, closing this issue.

closed 2020-11-18 by AlphaBetaPhi