facebookresearch/digit-design #18: CMOS OVM7692-RAAA and LEDs ARGB1313HS-TR are not available. Replacement possible?

b4be1, 2021-06-18:
Hi @AlphaBetaPhi,

I contacted several companies to produce the PCBs recently, and all of them unfortunately confirmed that it is not possible to get the camera and the LEDs at the moment. Is it possible to replace those parts with similar ones? If yes, could you recommend the suitable parts?

Best,
Boris

AlphaBetaPhi, 2021-06-18:
Hi @b4be1,

There is a wide shortage of components accross many different vendors. 

We have identified some replacements,

Camera - replace with OVM7692-RYAA
LED's - replace with 19-337C/RSBHGHC-A88/4T (this requires a design change to the flex lighting PCB)

New design for flex lighting PCB will be made available shortly.

Thank you,

Mike

b4be1, 2021-06-18:
Thank you very much for the quick reply. This is really good news!

b4be1, 2021-06-28:
Hi Mike,

I got a confirmation that the parts you suggested are all available. As I understood from your reply, the main PCB can be produced using the old design files by just substituting the camera, but for the flexible LED strip I should wait a bit for the new design files, right?

Best,
Boris

tkelestemur, 2021-06-28:
@b4be1 FYI, I was able to get a quote from Imagineering, Inc. with the original LED in the current BOM list and OVM7692-RYAA camera but they can't find the FT900Q-R sensor so I'm trying to find a substitute for them.

b4be1, 2021-06-28:
Thanks for the info @tkelestemur! You can replace FT900Q-R with FT900Q-C-T. I already produced the PCB with this newer microcontroller and everything works fine.

tkelestemur, 2021-06-28:
Yeah, that's what @AlphaBetaPhi told me to do. Where did you order your PCBs from? The manufacturer I'm talking to is saying it's gonna take 3 weeks and I'm wondering if I can find somewhere else that'll do it faster.

b4be1, 2021-06-28:
Yes, for me the waiting time is the same. I ordered from PCBWay.com

AlphaBetaPhi, 2021-06-28:
Hi @tkelestemur, @b4be1, unfortunately with this global shortage a lot of parts are out of stock. There is some available stock but most likely it would need to be purchased and shipped to the PCBA manufacturer (https://octopart.com/search?q=FT900Q-C-T&currency=USD&specs=0).

As for the flex design, there will be an update to this repo today. I'll be pushing the updated design.

AlphaBetaPhi, 2021-06-28:
Hi all, 

I have pushed rev. 2021-2 for the Digit Flex PCB to, [digit flex rev 2021-2](https://github.com/facebookresearch/digit-design/tree/master/hardware/fabrication/digit-flex/2021-2)

Additionally, for part swaps see [design notes](https://github.com/facebookresearch/digit-design#design-notes) for updates on part shortages and comments on acceptable part substitutions.

tkelestemur, 2021-06-29:
Thanks @AlphaBetaPhi

b4be1, 2021-06-29:
Many thanks @AlphaBetaPhi!

closed 2021-06-29 by b4be1

b4be1, 2021-06-29:
One small question: in [Digit BOM](https://github.com/facebookresearch/digit-design/blob/master/hardware/Digit%20BOM%20-%20rev%202020-1.pdf), the quantity of LED's is specified as 3 but there are 5 diodes D1-D5 in the schematic [Digit Flex Schematic](https://github.com/facebookresearch/digit-design/blob/master/hardware/Digit%20Flex%20Schematic%20-%20rev%202021-2.pdf). Is it a change in design and there are now 5 LEDs? That would mean I need to adjust the casing too, right?

AlphaBetaPhi, 2021-06-29:
Hi @b4be1, that's correct. There are 5 diodes in the schematic to allow for the addition of extra/optional lighting. The mechanical design is already accounting for these 2 extra LED's.

This is not currently supported at the moment but has been provisioned for future firmware releases.

b4be1, 2021-06-29:
Got it, thanks!