facebookresearch/digit-design #19: Doubts on the design of the acrylic clip

gabrielecaddeo, 2021-07-28:
Hi everyone,

thank you for the great sensor design that you made publicly available.

Me and my team reproduced the Digit and we noticed some incoherencies in the design of the mechanical components. We were wondering if the provided design files are correct and, in this case, we would like to highlight some issues we encountered.

The STL files of the acrylic clip (with or without snap) show a different representation of the component with respect to the one in the QuickStartGuide. As you can see in these pictures, the one in the Guide has some “windows” that prevent the noise due to a direct illumination of the camera by the LEDs light. The actual version in the STL files has not windows instead. Using the current design, generates unwanted artifacts in the output of the camera as you can see in the following picture. Artifacts due to the green and red LEDs are not visible in the output thanks to the presence of some protrusions in the main case.

[image]
[image]

Hence, we were wondering if that is normal or if there are some inconsistencies in the files in the git repository.

Thanks!

AlphaBetaPhi, 2021-08-10:
Hi @gabrielecaddeo,

Thank you for raising these issues. There will be an update shortly with the latest revision. 

This reflection should be solved with the updated acrylic clip file.

gabrielecaddeo, 2021-08-11:
Hi @AlphaBetaPhi,

Thanks for the answer. Looking forward to see the latest revision.

Thank you again.

AlphaBetaPhi, 2021-08-12:
Latest revision pushed, closing this issue.

closed 2021-08-12 by AlphaBetaPhi