facebookresearch/digit-design #20: Field of view reduced with respect to the gel size

gabrielecaddeo, 2021-07-28:
Hi everyone,

thank you for the great sensor design that you made publicly available.

Me and my team reproduced the Digit and we noticed some incoherencies in the design of the mechanical components. We were wondering if the provided design files are correct and, in this case, we would like to highlight some issues we encountered.

We tested the DIGIT and we had quite a good output. However, as it is possible to see in this video, the field of view seems reduced with respect to what were expecting. Indeed, even if the screw driver head is touching on the available surface of the gel, we cannot se its fingerprint on the output image.

[image]

Hence, we were wondering if that is normal or if there are some inconsistencies in the files in the git repository.

Thanks!

AlphaBetaPhi, 2021-08-10:
Hi @gabrielecaddeo,

Thank you for this issue and your teams interest in the project.

We have alternative designs with longer slightly longer housing the allow for full FOV but with some black edges due to non-square gel (the top of the finger tip)

Is this something that you would be interested in 3D printing?

gabrielecaddeo, 2021-08-11:
Hi @AlphaBetaPhi,

Thanks for the answer. Yes, we would be interested in the alternative design. We would be happy if you can share it with us, if that is not a problem.

Thank you again.