facebookresearch/digit-design #21: Sizes of screws do not match between main and back cases

gabrielecaddeo, 2021-07-28:
Hi everyone,

thank you for the great sensor design that you made publicly available.

Me and my team reproduced the Digit and we noticed some incoherencies in the design of the mechanical components. We were wondering if the provided design files are correct and, in this case, we would like to highlight some issues we encountered.

The back case and the main case are kept together by means of screws. In the STL files of the backhousing and of the main housing (both with or without snap) it is possible to see that the diameter of the screws does not match. The right screw for the back case is an M2 (and also for the main PCB), while the diameter of the main case suggests the use of an M4.
[image]

Hence, we were wondering if that is normal or if there are some inconsistencies in the files in the git repository.

Thanks!

AlphaBetaPhi, 2021-08-10:
Hi @gabrielecaddeo,

Thank you for raising this issue. An update with the latest revision will be pushed to the github repository. The designs are identical in terms of size but the mounting features are slightly different.

gabrielecaddeo, 2021-08-11:
Hi @AlphaBetaPhi, 

Thanks for the answer. Looking forward to see the latest revision.

Thank you again.

AlphaBetaPhi, 2021-08-12:
Hi @gabrielecaddeo, 

The latest mechanical revision has been pushed. Note, the new design contains open holes on the back of the center body enclosure. These will need to be covered by a black tape.

closed 2021-09-07 by AlphaBetaPhi