facebookresearch/digit-design #23: Can't install latest firmware [digit-2021-2a]

tkelestemur, 2021-08-19:
We have successfully built our first digit and started testing. After we ran the digit-programmer to install the firmware, we tried the examples in the digit interface. It works without a problem, however, in the output, the example warns that we need to use the latest firmware for independent RGB intensity control (see below). To install the latest firmware, we tried to flash the digit-2021-2a by running the following command: `python flash.py digit.firmware="../digit-2021-2a.bin"` but it failed with the following error:
```
(phd) ➜  digit-programmer git:(master) ✗ python flash.py digit.firmware="../digit-2021-2a.bin"
DIGIT programmer config:
INFO:DigitProgrammer:Attempting to find DIGIT serial mask in ../digit-2021-2a.bin binary.
INFO:DigitProgrammer:Found serial pattern...
INFO:DigitProgrammer:Serial number written to new firmware binary.
INFO:DigitProgrammer:Flashing DIGIT firmware...
Unplug and plug DIGIT into usb then press ENTER...
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available
INFO:DigitProgrammer:Finished flashing firmware to DIGIT!
```
Note that we installed the digit-interface from the source by cloning the master branch. The output of the demo_digit.py can be seen below. Even though, we currently don't need individual RGB intensity control, we wanted to let you know about this issues.
```
(phd) ➜  example git:(master) ✗ python demo_digit.py 
DEBUG:digit_interface.digit_handler:Finding udev devices with subsystem=video4linux, id_model=DIGIT
DEBUG:digit_interface.digit_handler:Following udev devices found: 
DEBUG:digit_interface.digit_handler:Device('/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/video4linux/video2')
DEBUG:digit_interface.digit_handler:Device('/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/video4linux/video3')
Connected DIGIT's to Host:
[{'dev_name': '/dev/video2',
  'manufacturer': 'Facebook',
  'model': 'DIGIT',
  'revision': '0101',
  'serial': 'D00000'},
 {'dev_name': '/dev/video3',
  'manufacturer': 'Facebook',
  'model': 'DIGIT',
  'revision': '0101',
  'serial': 'D00000'}]
DEBUG:digit_interface.digit:Digit object constructed with serial D00000
DEBUG:digit_interface.digit_handler:Finding udev devices with subsystem=video4linux, id_model=DIGIT
DEBUG:digit_interface.digit_handler:Following udev devices found: 
DEBUG:digit_interface.digit_handler:Device('/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/video4linux/video2')
DEBUG:digit_interface.digit_handler:Device('/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/video4linux/video3')
DEBUG:digit_interface.digit_handler:Searching for DIGIT with serial number D00000
INFO:digit_interface.digit:D00000:Connecting to DIGIT
INFO:digit_interface.digit:D00000:Setting stream defaults to QVGA, 60fps, maximum LED intensity.
DEBUG:digit_interface.digit:Default stream to QVGA {'width': 320, 'height': 240}
DEBUG:digit_interface.digit:D00000:Stream resolution set to 240w x 320h
DEBUG:digit_interface.digit:Default stream with 60 fps
DEBUG:digit_interface.digit:D00000:Stream FPS set to 60
DEBUG:digit_interface.digit:Setting maximum LED illumination intensity
WARNING:digit_interface.digit:You are using a previous version of the firmware which does not support independent RGB control, update your DIGIT firmware.
DEBUG:digit_interface.digit:D00000:LED intensity set to 0 (R: 0 G: 0 B: 0
Name: Left Gripper /dev/video2
	- Model: DIGIT
	- Revision: 101
	- CV Device?: True
	- Connected?: True
Stream Info:
	- Resolution: 320 x 240
	- FPS: 60
	- LED Intensity: 0
WARNING:digit_interface.digit:You are using a previous version of the firmware which does not support independent RGB control, update your DIGIT firmware.
DEBUG:digit_interface.digit:D00000:LED intensity set to 0 (R: 0 G: 0 B: 0
WARNING:digit_interface.digit:You are using a previous version of the firmware which does not support independent RGB control, update your DIGIT firmware.
DEBUG:digit_interface.digit:D00000:LED intensity set to 0 (R: 0 G: 0 B: 0
DEBUG:digit_interface.digit:D00000:Stream resolution set to 240w x 320h
DEBUG:digit_interface.digit:D00000:Stream FPS set to 30
Frame WxH: 320240
```

lianghongzhuo, 2021-08-19:
Hi, did you try unplugging the digit before run flash.py, after the code suggests you plug in the digit device, plug in it?

tkelestemur, 2021-08-19:
Yes, that's how we flashed the default firmware. I also wait for a few seconds until I see the digit in the output of `lsusb`.

lianghongzhuo, 2021-08-19:
maybe because the device was flashed to a firmware, then the idVendor and idProduct have changed?
Is your `lsusb` output indicate your device as 0403:0fde or 2833:0209?

tkelestemur, 2021-08-19:
It's `2833:0209`

lianghongzhuo, 2021-08-19:
I don't have the chip in my hand right now to verify my guess, but I think when the id becomes ` 2833:0209` computer recognizes it as a digit sensor, and you can't flash firmware anymore.
Maybe try to flash the board when it still recognizes as ` 0403:0fde`. In 3 seconds after you plugin, it should still recognize as  `0403:0fde`

tkelestemur, 2021-08-19:
You were right, after I replug it, the switches from `0403:0fde` to `2833:0209`. I pressed enter before it changes and I think I was able to flash the new firmware. At least, the example doesn't throw the warning anymore. Here is the output of the flash.py:
```
(phd) ➜  digit-programmer git:(master) ✗ python flash.py digit.firmware="../digit-2021-2a.bin" digit.serial=0       
DIGIT programmer config:
INFO:DigitProgrammer:Attempting to find DIGIT serial mask in ../digit-2021-2a.bin binary.
INFO:DigitProgrammer:Found serial pattern...
INFO:DigitProgrammer:Serial number written to new firmware binary.
INFO:DigitProgrammer:Flashing DIGIT firmware...
Unplug and plug DIGIT into usb then press ENTER...
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0403:0fde
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 256
Copying data from PC to DFU device
Download	[=========================] 100%        44484 bytes
Download done.
Sent a total of 44484 bytes
state(6) = dfuMANIFEST-SYNC, status(0) = No error condition is present
dfu-util: unable to read DFU status after completion
dfu-util: can't detach
Resetting USB to switch back to runtime mode
dfu-util: error resetting after download
INFO:DigitProgrammer:Finished flashing firmware to DIGIT!

lianghongzhuo, 2021-08-19:
Glad to hear that!

lianghongzhuo, 2021-08-20:
So, is your problem solved or not?

AlphaBetaPhi, 2021-08-20:
Hi all,

Thank you for your interest in the project.

When flashing an initial device, with the latest firmware `2a`, the `-padded.dfu` should be used initially. After this, the DIGIT will accept firmware upgrades without padding. Additionally, the firmware can be re-programmed, or downgraded using the same utility even after the vendor and device ID change.

The device should be power cycled after flashing the firmware.

closed 2021-08-20 by AlphaBetaPhi