facebookresearch/digit-design #27: Replacing the FT900Q-x with a FT901/2/3Q-x

dornik, 2021-09-05:
Hi everyone, hi @AlphaBetaPhi, 

Thanks for sharing your work!

Is it possible to replace the FT900Q-x with another microcontroller of the FT90x series? From the [datasheet](https://docs.rs-online.com/c058/0900766b816a4aa0.pdf) (part numbers on page 3 and pin configurations on pages 8ff) it looks as if the only difference would be the missing Ethernet/CAN bus support on the FT901/2/3Q-x.

Thanks in advance!

AlphaBetaPhi, 2021-09-07:
Hi @dornik,

Thank you for your interest in the project.

It is possible to replace the FT900Q with the 901/2/3 variant. Please note, use the C revision of the silicon - FT90X-Q-C.

dornik, 2021-09-07:
Cool, I'm glad that this works and we will make sure to use the C revision.

Thanks for your help!

closed 2021-09-07 by dornik