facebookresearch/digit-design #28: Firmware versions

b4be1, 2021-09-15:
What firmware version should one use?

By default, `digit-2020-1b-dfu.bin` is specified

https://github.com/facebookresearch/digit-design/blob/ead7143a6b2e728dde39d9a18e598969253a2efe/firmware/digit-programmer/programmer.yaml#L3

However, I am getting an error

```
Download        [====================== ]  98%        254464 bytes failed!
state(10) = dfuERROR, status(8) = Cannot program memory due to received address that is out of range
dfu-util: can't detach
Resetting USB to switch back to runtime mode
```

If I use `digit-2020-1b.bin`, then everything is working fine.

With `digit-2021-2a.bin`, I can install it, but then upon connecting to the Digit, the red LED turns off.

With `digit-2021-2a-padded.bin`, I am getting the same download error as above at 96%.

AlphaBetaPhi, 2021-09-17:
Hi @b4be1,

Which version of the digit-interface are you using? Can you pull the latest from the digit-interface repository and try?

b4be1, 2021-09-18:
Hi @AlphaBetaPhi,

Thanks for the explanation. I used the latest version of the digit-interface. I ran the default `-dfu.bin` first (probably successfully, I didn't pay attention) and then tried the other firmware versions. It makes sense then that when I install `-dfu.bin` for the second time, it does not terminate successfully, because it is supposed to be installed only in the beginning.

The only unexpected behavior is then that the red LED turns off when I use `digit-2021-2a.bin`. It does not bother me too much at the moment though, because I can use the other version of the firmware.

AlphaBetaPhi, 2021-09-20:
Hi @b4be1 ,

Can you post the output of

```
from digit_interface.digit_handler import DigitHandler

digits = DigitHandler.list_digits()
```

and then post the following after connecting to that DIGIT using,

```
from digit_interface.digit import Digit

d = Digit("Dxxxxx") # your serial number 
d.connect()
print(d.info())
```

b4be1, 2021-09-21:
Hi @AlphaBetaPhi,

I started with a fresh Digit. I first installed `digit-2020-1b-dfu.bin` and then ran the commands you provided. Here is the output

> [{'dev_name': '/dev/video0', 'manufacturer': 'Facebook', 'model': 'DIGIT', 'revision': '0101', 'serial': 'D00001'}, {'dev_name': '/dev/video1', 'manufacturer': 'Facebook', 'model': 'DIGIT', 'revision': '0101', 'serial': 'D00001'}]
> Name: None /dev/video0
>         - Model: DIGIT
>         - Revision: 0101
>         - CV Device?: True
>         - Connected?: True
> Stream Info:
>         - Resolution: 320 x 240
>         - FPS: 60
>         - LED Intensity: 255

Then I installed `digit-2020-1b.bin` and ran the commands. The output was the same. After that, I installed `digit-2021-2a.bin` and got the following output

> [{'dev_name': '/dev/video0', 'manufacturer': 'Facebook', 'model': 'DIGIT', 'revision': '0200', 'serial': 'D00001'}, {'dev_name': '/dev/video1', 'manufacturer': 'Facebook', 'model': 'DIGIT', 'revision': '0200', 'serial': 'D00001'}]
> Name: None /dev/video0
>         - Model: DIGIT
>         - Revision: 0200
>         - CV Device?: True
>         - Connected?: True
> Stream Info:
>         - Resolution: 320 x 240
>         - FPS: 60
>         - LED Intensity: 255

As I said before, the only noticeable difference between `2020-1b` and `2021-2a` is that the red LED stops shining when I run `d.connect()`. The camera stream seems to be OK.

I furthermore did some additional experiments on another fresh Digit. I first installed `2021-2a-padded.bin`. This worked fine. So, basically, I am able to either install `2020-1b-dfu.bin` or `2021-2a-padded.bin` as the initial firmware. After that, I am able to install `2020-1b` or `2021-2a`, but I can't install `-dfu` or `-padded` versions anymore because they seem to run out of memory at around 96%–98%.

One thing is still unclear to me: is there a difference between `2020-1b` and `2020-1b-dfu`? It seems all basic functionality is already included in `2020-1b-dfu`.

Some parts had to be replaced in the production of these Digits. I think even the LEDs are different. I attached the materials used in the Digits I am testing on:
[LEDs.pdf](https://github.com/facebookresearch/digit-design/files/7201612/LEDs.pdf)
[PCB.pdf](https://github.com/facebookresearch/digit-design/files/7201614/PCB.pdf)

AlphaBetaPhi, 2021-09-22:
Hi @b4be1,

It looks like the correct firmware is installed, but the digit-interface library version has not been updated as the `LED intensity` is `255`. On new `digit-interface` library this will set the LED intensity to a 12-bit value. 

Can you try upgrading the pip package,

`pip install digit-interface --upgrade` and try with the new firmware.

b4be1, 2021-09-23:
Hi @AlphaBetaPhi,

I was on version `0.1.4`. After updating to `0.2.1` everything works fine when using `digit-2021-2a.bin`. LED intensity is 4095 in the test output. Thanks!

closed 2021-09-23 by b4be1