facebookresearch/digit-design #31: device not recognized after flashing: "Device not responding to setup address."

roberttamas98, 2021-11-17:
Flashing the firmware on the Digit doesn't seem to work for me, I tried on different PCs and with different Digits, but got the same result as follows:

Before flashing the Digit and using the following command:
    dmesg | grep usb

I get my latest plugged in usb-device and the microcontroller (FT900) is recognised:
    [3712107.985896] usb 1-7: USB disconnect, device number 120
    [3712113.118788] usb 1-7: new high-speed USB device number 121 using xhct_hcd
    [3712113.268139] usb 1-7: New USB device found, idVendor=0403, idProduct=ofde
    [3712113.268144] usb 1-7: New USB device strings: Mr=1, Product=2, SerialNumber=3
    [3712113.268148] usb 1-7: Product: FT900
    [3712113.268150] usb 1-7: Manufacturer: FTDI
    [3712113.2681537 usb 1-7:
    SerialNumber:00000000

after flashing (following all the the instructions):
    python3 flash.py digit.serial=45

and getting following output:
    INFO:DigitProgrammer:DIGIT programmer config: {'digit': {'serial': 45, 'firmware': '../digit-2020-1b-dfu.bin'}, 'usb': {'idVendor': 10291, 'idProduct': 521}}
    INFO:DigitProgrammer:Attempting to find DIGIT serial mask in ../digit-2020-1b-dfu.bin binary.
    INFO:DigitProgrammer:Found serial pattern...
    INFO:DigitProgrammer:Serial number written to new firmware binary.
    INFO:DigitProgrammer:Flashing DIGIT firmware...
    Unplug and plug DIGIT into usb then press ENTER...
    dfu-util 0.9

    Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
    Copyright 2010-2016 Tormod Volden and Stefan Schmidt
    This program is Free Software and has ABSOLUTELY NO WARRANTY
    Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

    dfu-util: DFU suffix CRC does not match
    dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
    dfu-util: No DFU capable USB device available
    INFO:DigitProgrammer:Finished flashing firmware to DIGIT!

something obviously went wrong, but no idea what and if I try to list all the usb devices again:
    dmesg | grep usb

it doesn't recognize the microcontroller  (FT900) either anymore:
    [ 1766.233552] usb usb1-port6: attempt power cycle
    [ 1766.645491] usb 1-6: new full-speed USB device number 67 using xhci_hcd
    [ 1766.645743] usb 1-6: Device not responding to setup address.
    [ 1766.853783] usb 1-6: Device not responding to setup address.
    [ 1767.061495] usb 1-6: device not accepting address 67, error -71

This was happening with two different digits. After manufacturing another and confirming by "dmesg | grep usb" that the microcontroller is being recognized and before flashing again I also tried:
    sudo dfu-util -l

and got the following output:
    dfu-util 0.8
    Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
    Copyright 2010-2014 Tormod Volden and Stefan Schmidt
    This program is Free Software and has ABSOLUTELY NO WARRANTY
    Please report bugs to dfu-util@lists.gnumonks.org
    Found DFU: [0403:0fde] ver=2300, devnum=93, cfg=1, intf=0, alt=0, name="UNKNOWN", serial="00000000"

Does someone else had this issue and got a solution?

AlphaBetaPhi, 2021-11-17:
Hi @roberttamas98,

Thank you for your interest in the project. A few questions:

* Did you manufacture these DIGIT's yourself?
* What MFG part number was used for the FT900, can you list the full part number?

roberttamas98, 2021-11-18:
Thanks for the quick answer
Yes I'm manufacturing them myself.
I have the FT90[1/2/3]-Q-C, but I only used the FT902Q until now.

AlphaBetaPhi, 2021-11-18:
Are you using the FT902Q or FT902Q-C on the devices you assembled?

roberttamas98, 2021-11-18:
I'm using the FT902-Q-C

AlphaBetaPhi, 2021-11-20:
Hi @roberttamas98,

What is the output of `lsusb -vvv' does the DIGIT DFU mode show up?

Can you try this command,

```
$ python flash.py digit.firmware="../digit-2021-2a-padded.bin" digit.serial=123
```

roberttamas98, 2021-11-26:
As I was already suspecting and you suggested, I was using the wrong firmware, but it works now with your command. Is there a possibility to save those digits which where flashed using the wrong firmware?

roberttamas98, 2021-12-06:
Is there someone else experiencing similar issues with flashing the firmware or does someone know if digits where the flashing somehow failed can be saved?

AlphaBetaPhi, 2021-12-07:
@roberttamas98 If you've already flashed the DIGIT and it enumerates as "FT900 DFU Mode" then try flashing with

```
python3 flash.py digit.firmware="../digit-2021-2a.bin" digit.serial=2
```

We will be releasing an updated programmer which signs binaries to resolve this.

roberttamas98, 2021-12-07:
unfortunately it still doesn't seem to work:

python3 flash.py digit.firmware="../digit-2021-2a.bin" digit.serial=2
INFO:DigitProgrammer:DIGIT programmer config: {'digit': {'serial': 2, 'firmware': '../digit-2021-2a.bin'}, 'usb': {'idVendor': 10291, 'idProduct': 521}}
INFO:DigitProgrammer:Attempting to find DIGIT serial mask in ../digit-2021-2a.bin binary.
INFO:DigitProgrammer:Found serial pattern...
INFO:DigitProgrammer:Serial number written to new firmware binary.
INFO:DigitProgrammer:Flashing DIGIT firmware...
Unplug and plug DIGIT into usb then press ENTER...
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0403:0fde
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 256
Copying data from PC to DFU device
Download	[=========================] 100%        44484 bytes
Download done.
Sent a total of 44484 bytes
state(6) = dfuMANIFEST-SYNC, status(0) = No error condition is present
dfu-util: unable to read DFU status after completion
dfu-util: can't detach
Resetting USB to switch back to runtime mode
dfu-util: error resetting after download
INFO:DigitProgrammer:Finished flashing firmware to DIGIT!

I guess I'll wait for the updated programmer and try to work with the one that somehow worked, but thanks for the help for now

AlphaBetaPhi, 2021-12-07:
@roberttamas98, could you contact me via lambetam [at] fb [dot] com

Let's see if we can root cause this and post the solution on this issue.

csufangyu, 2021-12-15:
hi, can you solve it?I get the same problem.

roberttamas98, 2021-12-22:
In the live-call with mike, we concluded that those problems are most certainly/definitely hardware related issues.

the following outputs to the dmesg command point to different hardware problems:

[  501.072897] usb 3-6: New USB device strings: Mfr=1, Product=3, SerialNumber=2
[  501.072898] usb 3-6: Product: FT900 DFU Mode
[  501.072899] usb 3-6: Manufacturer: FTDI
[  501.072900] usb 3-6: SerialNumber: FT424242
[  503.671422] usb 3-6: USB disconnect, device number 6
[  635.055100] usb 3-6: new full-speed USB device number 7 using xhci_hcd
[  635.207915] usb 3-6: New USB device found, idVendor=0403, idProduct=0fde, bcdDevice=23.00

most likely a problem with the camera, either a short, no contact to a pin or maybe a broken camera

[  641.686334] usbcore: registered new interface driver uvcvideo
[  641.686335] USB Video Class driver (1.1.1)
[  924.223229] usb 3-6: USB disconnect, device number 8
[  935.020330] usb 3-6: new full-speed USB device number 9 using xhci_hcd
[  935.148318] usb 3-6: device descriptor read/64, error -71
[  935.384328] usb 3-6: device descriptor read/64, error -71
[  935.620302] usb 3-6: new full-speed USB device number 10 using xhci_hcd
[  935.752329] usb 3-6: device descriptor read/64, error -71
[  935.988338] usb 3-6: device descriptor read/64, error -71
[  936.096337] usb usb3-port6: attempt power cycle
[  936.524311] usb 3-6: new full-speed USB device number 11 using xhci_hcd
[  936.524521] usb 3-6: Device not responding to setup address.
[  936.732539] usb 3-6: Device not responding to setup address.
[  936.944282] usb 3-6: device not accepting address 11, error -71
[  937.072313] usb 3-6: new full-speed USB device number 12 using xhci_hcd
[  937.072530] usb 3-6: Device not responding to setup address.
[  937.280522] usb 3-6: Device not responding to setup address.
[  937.488310] usb 3-6: device not accepting address 12, error -71
[  937.488363] usb usb3-port6: unable to enumerate USB device

when not recognized at all it might be due to some problem with the contatcs of the microprocessor which can't be seen from the side

[  641.611449] usb 3-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  641.611449] usb 3-6: Product: DIGIT
[  641.611450] usb 3-6: Manufacturer: Facebook
[  641.611451] usb 3-6: SerialNumber: D00123
[  641.685688] uvcvideo: Found UVC 1.10 device DIGIT (2833:0209)
[  641.686077] uvcvideo: Failed to set UVC probe control : -32 (exp. 34).
[  641.686269] input: DIGIT: DIGIT as /devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6:1.0/input/input23

and the last is for reference, how it looks for a working digit.

Thanks again Mike :)

pekkykang, 2022-07-10:
I have the same problem, and tested four  PCB with the dmesg command. All got:
[ 4417.712816] usb 1-13: New USB device strings: Mfr=1, Product=3, SerialNumber=2
[ 4417.712822] usb 1-13: Product: FT900 DFU Mode
[ 4417.712826] usb 1-13: Manufacturer: FTDI
[ 4417.712830] usb 1-13: SerialNumber: FT424242
[ 4420.153908] usb 1-13: USB disconnect, device number 112
[ 4422.015339] usb 1-9: new full-speed USB device number 113 using xhci_hcd
[ 4422.164651] usb 1-9: New USB device found, idVendor=0403, idProduct=0fde, bcdDevice=23.00

Does these means I got four broken PCB?

pekkykang, 2022-07-10:
Will these be possible that all four cameras are bad?

AlphaBetaPhi, 2022-07-10:
@pekkykang Did you power cycle the DIGIT?

pekkykang, 2022-07-11:
I think the only thing I can do is change the camera supplier and PCB manufacturer.

AlphaBetaPhi, 2022-07-11:
Hi @pekkykang,

This means that the device has been programmed successfully if the LED's are turning on. If the device does not enumerate as a DIGIT after the LED's are on it means it cannot find or configure the camera. Did you assemble these yourself?

pekkykang, 2022-07-11:
> Hi @pekkykang,
> 
> This means that the device has been programmed successfully if the LED's are turning on. If the device does not enumerate as a DIGIT after the LED's are on it means it cannot find or configure the camera. Did you assemble these yourself?

Hi, Phi:

Yes, I assemble these myself. I think I will change the camera supplier and PCB manufacturer.

AlphaBetaPhi, 2022-07-11:
Which camera model number did you use? Do you have any pictures of how you assembled the devices? 

If they're hand soldered, I would suggest reflowing the camera once more.

pekkykang, 2022-07-11:
I use OVM7692-RAAA and ask a PCB manufacturer to use SMT to do it
[image]
.

AlphaBetaPhi, 2022-07-11:
Can you take a close up picture of one of the PCBs?

pekkykang, 2022-07-11:
> Can you take a close up picture of one of the PCBs?
[image]

AlphaBetaPhi, 2022-07-11:
It looks like the camera has been soldered on incorrectly. The orientation is wrong. This would explain why the camera the device is not booting up.

pekkykang, 2022-07-11:
> It looks like the camera has been soldered on incorrectly. The orientation is wrong. This would explain why the camera the device is not booting up.

Thank you. This helps me very much. I will complain to the manufacturer and ask them to give me the money back😂

AlphaBetaPhi, 2022-07-11:
You could remove the camera with hot air and then correct the orientation, it needs to be rotated 180deg.

pekkykang, 2022-07-11:
> You could remove the camera with hot air and then correct the orientation, it needs to be rotated 180deg.

OK

closed 2022-07-11 by AlphaBetaPhi