facebookresearch/digit-design #40: Artefacts in Gel Adhesion Layers within Spec?

elishafer, 2024-02-04:
I recently bought a few Digit sensors and replacement gels from Gelsight.
I was surprised that some of the gels had dust specs, bubbles and pieces of white silicon in the adhesion layer between the acrylic and clear silicone.
Gelsight said the sample photos I sent them are within the QC spec. How would artefacts like these in the following pictures affect inference? What is the QC spec for artefacts in the adhesion/silicon layers?
[image]
[image]
[image]
[image]

robertocalandra, 2024-02-05:
Hi @elishafer,

Yes, that is indeed a very clear manufacturing issue with the gel.

Unfortunately, you are not the only person to be having issues with the QC of GelSight Inc recently (https://github.com/facebookresearch/digit-interface/issues/15), but there is nothing I can do about it.

Good news is that if you use machine learning to learn models, I don't expect imperfections in the silicone to strongly impact the performance of your model.

elishafer, 2024-02-05:
HI @robertocalandra,
Thank you for your prompt answer. I totally understand that Gelsight is a separate entity and it's nothing to do on your part

> Good news is that if you use machine learning to learn models, I don't expect imperfections in the silicone to strongly impact the performance of your model. 
That's a relief 

Do you happen to have the QC manufacturing spec for the gel so that I can negotiate with GelSight for a replacement? 
I have a feeling that the following gel replacements shouldn't be in spec:

[image]
[image]